exelban/stats #1179: Hottest and average CPU sensor is missng

rajtslegr, 2022-11-16:
Hottest and average CPU sensor is missing in the list.

**Details:**
 - Device: MacBook Air M1
 - macOS: 13.0.1 (22A400)
 - Application version: 2.7.44

exelban, 2022-11-16:
Are you sure you have v2.7.44?

rajtslegr, 2022-11-16:
[image]

[image]
[image]

exelban, 2022-11-16:
Could you please try [this](https://github.com/exelban/stats/releases/tag/v2.7.42) version?

rajtslegr, 2022-11-16:
> Could you please try [this](https://github.com/exelban/stats/releases/tag/v2.7.42) version?

All good with .42, it started with .43 i think.

exelban, 2022-11-16:
hmm, strange

exelban, 2022-11-16:
because in the latest release there are these sensors.

exelban, 2022-11-16:
Could I send you a debug version to see what is missing?

exelban, 2022-11-16:
there could be some bug with HID sensors. I have no m1 and cannot check it(
Thanks, I will post a debug build later today.

rajtslegr, 2022-11-16:
I just updated to 2.7.44 from 2.7.42 (your upload in this issue) and it miraculously works 🤣

closed 2022-11-16 by rajtslegr