exelban/stats #1180: APFS volume capacity ignored

michalcs, 2022-11-16:
I've noticed my APFS volumes weren't showing the correct free space left. Looked at it closely and realised the APFS Capacity seems to be ignored

### Steps

1. Get an APFS drive with an existing Container
2. In Disk Utility right-click that Container and `Add APFS volume`
3. Select `Size Options`
4. Set `Quota Size` to `20GB` and `OK`
5. Observe the disk/volume size and free space in Disk Utility, Finder and Stats

### Expected

The volume should show `20GB` size and base the free space calculations on that

### Actual

Stats seems to ignore the set APFS capacity and calculates the free space based on the parent container

[image]

exelban, 2022-11-17:
Could you check your `Macintosh HD` format?

michalcs, 2022-11-17:
[image]

[diskutil_list.txt](https://github.com/exelban/stats/files/10030746/diskutil_list.txt)
[diskutil_apfs_list.txt](https://github.com/exelban/stats/files/10030747/diskutil_apfs_list.txt)

exelban, 2022-11-17:
so its apfs and Stats show the capacity?

michalcs, 2022-11-17:
> so its apfs and Stats show the capacity?

- `Macintosh HD` = APFS volume and Stats show the correct capacity
- `Coconut` = APFS volume and Stats ignore the volume capacity, show the container capacity instead

closed 2022-11-20 by exelban