exelban/stats #1184: Network Element is showing false stats

seppel3599, 2022-11-18:
When Downloading something (constant download) the widget is not showing the correct throughput (see screenshot).

- My Machine: MacBook Pro 14" with M1 Pro running macOS 13.0.1 (22A400)
- Version of Stats: 2.7.44
- What I tried: App Restart, Mac Restart, Checking for Updates in macOS and Stats-App

[image]

exelban, 2022-11-18:
Please share the screenshot of the settings.

seppel3599, 2022-11-20:
Of course: 
[image]
[image]

Do you need more?

exelban, 2022-11-20:
Nope.

closed 2022-11-20 by exelban

seppel3599, 2022-11-28:
Hello,
thank you for checking.
But with a threshold of "50 KB", a throughput of 1.3 MB should be displayed, or am I missing something?