exelban/stats #1187: no bug, more suggestion

aarontemp123, 2022-11-20:
In the network module, it should display the information you get when you control + click the wifi icon.
things missing like router ID or the TX rate.

exelban, 2022-11-20:
What is the router id?

aarontemp123, 2022-11-20:
Router ID is another name for default gateway, handy if you need to troubleshoot something.
TX rate is useful because it's shows the theoretical max of your wifi speed, you might have full bars but only be connected to a wifi b/g/n which will be a lot slower than AC/AX, your TX rate will variate depending on the width of the channel that is being used: 20,40,80,160 MHz.
[image]

exelban, 2022-11-20:
Ok, I will add that.

closed 2022-11-27 by exelban