exelban/stats #1189: Since a few days the marker of the free space of the SSD is not correct

ToniCurtis66, 2022-11-20:
A few days ago the SSD free space indicator is wrong...
[image]

exelban, 2022-11-20:
What version of Stats do you have?

exelban, 2022-11-21:
Hmm, please try [this](https://github.com/exelban/stats/releases/tag/v2.7.42) version.

ToniCurtis66, 2022-11-21:
Work fine!
Thx.

exelban, 2022-11-21:
Thanks for reporting. Will try to fix that.

ToniCurtis66, 2022-11-21:
Ok, thanks.

exelban, 2022-11-23:
Hi. And there is a problem. There are 2 APIs to receive a free space. And in the `v2.7.42` I changed the order of using them. (if the first is not returned the data use the second method) It was done because starting from the last update of macOS API that was used previously creates a lot of nested logs. Disabling this API also allows for decreasing the CPU usage of this module.

And the problem now is that there is no solution right now. Also, the API that is used now does not show the wrong free space. But it just does not include purging space (`space expected to be cleared by purging non-essential and cached resources`).

TL.DR: I have no good solution at this moment. Maybe it will be fixed in the next release of macOS, and then I could return the previous API.

ToniCurtis66, 2022-11-23:
Good. Well, for now I'm leaving version 2.7.42 installed. Thanks for the
explanation.
Greetings.

ToniCurtis66, 2023-01-14:
Hi
The version 2.8.3 work fine!
I am editor of an important Spanish blog about Mac and Apple. Since I have
been using Stats for a long time and I like it a lot, I have written an
article about this app.
I hope you like it.
https://www.soydemac.com/controla-el-estado-de-tu-mac-permanentemente-con-stats/

exelban, 2023-01-15:
Hi. Thanks for the feedback.
Unfortunately, it's still not fixed. macOS still have strange behavior when using the correct API.
The reason why you see the correct size it's more probably because there is purgeable data in the system.

I monitor that issue from time to time. I will give to know when it will be fixed.

ToniCurtis66, 2023-01-15:
Perfect. At the moment on my iMac 24" it works fine...
I await your news with the final arrangement.

exelban, 2023-01-29:
In the new macOS update the problem is still present. I will look for a new way to obtain a free space.

exelban, 2023-02-06:
Hi. Unfortunately looks like Apple has no plans to fix that. The old way of fetching the available purged space cannot be used because causing high CPU usage. Right now the value that shows in the Stats is correct but just does not include the purgeable space. I cannot find a way to obtain purgeable space for the disk.

closed 2023-02-06 by exelban

ToniCurtis66, 2023-02-06:
2.8.7 works fine!

nishyp, 2023-03-24:
Hi, I'm also facing the same problem
MacOS Catalina 10.15.7
Stats Version 2.8.13

[image]