exelban/stats #1193: Stats shows network as down when using router-based VPN

rickogorman, 2022-11-24:
**Describe the bug**
I am using a VPN that runs in my ASUS router. When I connect to the internet via the VPN (using a guest network) the icon for network shows as red, whereas it is green for standard/non-VPN router networks. The VPN/guest network is working fine however, so the network is not down (I'm actually connected to GitHub currently using the VPN). The pop-out menu (when I click the icon in menu bar) is working, showing external and internal IPs, status as UP, etc., but internet connection as down.

**Details:**
macbook air M1
macOS 12.6.1 (monterey)
Stats 2.7.45
[image]

exelban, 2022-11-24:
What is the result of `ping 1.1.1.1`?

rickogorman, 2022-11-25:
I've attached the results for standard (non-VPN) connection and the VPN connection. The standard is working fine, the VPN one is timing out. But I can use the web fine. Possibly the VYN is blocking 1.1.1.1 or something?

[image]
[image]
[image]

exelban, 2022-11-25:
Yes, your VPN for some reason blocks Cloudflare IP (which is the default in the Stats for ICMP). You can change the ICMP host in the Network module settings.

closed 2022-11-25 by exelban

rickogorman, 2022-11-25:
Thanks Serhiy. It is likely that the VPN will block all such DNS servers except their own, I would guess, to prevent inadvertent leakage. 

Rick