exelban/stats #1194: [Feature Request] View usage history in expanded window

iasontran, 2022-11-26:
Hi,

Not sure how feasible this is but I figured I would throw a suggestion out anyways. One feature I miss from iStat Menus is the ability to highlight my cursor on the usage history section of the CPU, RAM, or GPU and get an expanded view that would allow me to get an overview of the usage over various timespans such as the past 24 hours, week, month, etc.

This would probably be a lengthy feature to implement but I wanted to ask if it's possible at some point in the future.

exelban, 2022-11-26:
Hi. Thanks for the feedback.
Yeap, I have the plan to add this one day.

victor-marino, 2023-12-26:
Hi all.

Just a suggestion to further improve this feature.

I think it would be great if the "connectivity history" could also be plotted in the historical traffic chart itself (both the small one and the new, bigger one).

For instance, during the time period when the connection was down, we could change the background color of the chart to be a translucent shade of whatever "Non-Active" color we have picked (e.g.: red shade).

This could even remove the need to have a dedicated "connectivity history" chart like we do now, while also making it much easier to correlate the traffic data with the downtime data, since both would be plotted along the same timeline now.

What do you think?

ChillarAnand, 2024-05-30:
Currenlty, I am using glances + Influxdb + Grafana to see historical data. I am eagerly waiting for this. 

I just noticed that this was moved to "In Progress". If you need help with testing, let me know. @exelban