exelban/stats #1195: Permission request popup everytime selecting a fan control option

Jerry23011, 2022-11-27:
**Describe the bug**
Whichever option I click on the fan controlling option will cause the permission requestion popup appear asking me for password, and the fan control doesn't work, even if I set a value under manual mode or simply click the cooling option. 

**Details:**
 - Device: MacBook Pro 14'
 - macOS: 12.6.1
 - Application version: v2.7.46

 Screenshots below
1. Under the sensors popup
[image]
2. Permission is required whichever option I click.
[image]

exelban, 2022-11-27:
Hi. Please check if there is a `eu.exelban.Stats.SMC.Helper` inside `/Library/PrivilegedHelperTools` after you install the helper.

exelban, 2022-11-27:
And to ensure. Are you giving permission to install the helper or not?

Jerry23011, 2022-11-27:
> Hi. Please check if there is a `eu.exelban.Stats.SMC.Helper` inside `/Library/PrivilegedHelperTools` after you install the helper.

Yes, it's there

> And to ensure. Are you giving permission to install the helper or not?

Yes, I did, gave permission every time it pops up.

Jerry23011, 2022-11-27:
Just found something interesting, when the mode is Automatic and I click the "❄️" icon, the popup window will appear twice while it pops up only once in all the other situations.

exelban, 2022-11-28:
Hmm. It's strange. I need to make a debug build to understand what is going on there. Would you like to help with that?

Jerry23011, 2022-11-28:
Of course, no problem

exelban, 2022-11-28:
[Stats.dmg.zip](https://github.com/exelban/stats/files/10103515/Stats.dmg.zip)

exelban, 2022-11-28:
Please run this build and try to manipulate the fans. In the `Documents` folder you will find a file `log.txt`. Please share this file after a run.

exelban, 2022-11-28:
Could you please check if there is a `"eu.exelban.Stats.SMC.Helper" => enabled` in the output of the `launchctl print-disabled system`?

exelban, 2022-11-28:
You can try to enable it via `sudo launchctl enable system/eu.exelban.Stats.SMC.Helper`.
But I'm not sure it will help.

Jerry23011, 2022-11-28:
Now it works! Thanks a lot! Surprised that it's the lauching item's problem.

Jerry23011, 2022-11-28:
I will now test v2.7.46 and see if it works

exelban, 2022-11-28:
Looks like at some moment you have blocked the helper.

Jerry23011, 2022-11-28:
> Looks like at some moment you have blocked the helper.

Yes indeed, v2.7.46 works perfectly fine, thank you very much 🙏🏻

closed 2022-11-28 by Jerry23011