exelban/stats #1196: The temperature notification icon broke

pedropra, 2022-11-27:
**Describe the bug**

Hello,  I installed the Status App and the temperature icon is not showed correctly 

[image]

**Details:**
 - Device:  Macbook Pro 2018
 - macOS: Ventura 13.0.1 
 - Application version: 2.7.46

Jerry23011, 2022-11-27:
Hi, which temperature sensor are you referring to specifically?

pedropra, 2022-11-27:
Hi, all this. 
[image]

Jerry23011, 2022-11-27:
Actually I mean one of these; I just tested all of them
[image]

closed 2022-11-27 by exelban

pedropra, 2022-11-27:
Works now! Thank you my friend.

Jerry23011, 2022-11-27:
Awesome