exelban/stats #1199: [Feature Request] Wifi icon change in case of wired network or wifi network

BetterB0y, 2022-11-28:
I would really appreciate this feature.

exelban, 2022-12-04:
Hi. Thanks for the feedback.
Do you have an idea of how it could be indicated in the menu bar?

BetterB0y, 2022-12-04:
There is an app: https://apps.apple.com/us/app/ethernet-status/id1186187538?mt=12
It could be something like this.

BetterB0y, 2022-12-04:
But it should also have if ethernet state is not connected check if wifi is connected and show appropriate icon

exelban, 2022-12-04:
Hmm, it requires a new widget. Now I'm working on an app redesign. Maybe after that, I will return to that feature. I cannot promise that it will be implemented. But if I will have a free time I will take a look at this.

exelban, 2023-06-02:
Sorry, but it will not be added. I have no time for this widget.

closed 2023-06-02 by exelban