exelban/stats #1200: [Feature Request] Synchronize Network upload/download widget color to graph color

fiveangle, 2022-11-30:
Is it possible to have the Upload and Download colors of the menubar widget (arrows, dots, values, etc) to match those set in the preferences for the pop-up graph ? This way those with different themes of menubar colors can set their best/favorite contrast colors.

[image]

Thank you for your consideration !

fiveangle, 2022-11-30:
BTW- Thank you again for allowing us the feature to color the upload/download values rather than use a pictogram (#936)

I've just donated to the project and want you to know we value your effort. Regardless if you implement this feature or not, your utility is wonderful.

exelban, 2022-11-30:
Hi. Thanks for the feedback and support.
Hmm, yeap. I will match the color of the value to the pictogram's color.

closed 2022-11-30 by exelban

fiveangle, 2022-12-11:
OMG, i'm an idiot. I'd not scrolled down that far I suppose ! Sorry for the noise ! And thanks again for such a great utility ! ❤️