exelban/stats #1203: no cpu temp sensor on M1 MBP 13" 2020

georgedimac, 2022-12-02:
**Details:**
 - Device: [MBP 13" 2020 M1 17,1]
 - macOS: [12.6.1]
 - Application version: [e.g. 2.7.46]

It does not interpret cpu temp sensors, also i don't know which one is on this machine because it's different than regular Tp0d on m1pro 2021 machines.

[image]
[image]

closed 2022-12-02 by exelban

georgedimac, 2022-12-02:
can you please give an answer? you closed with no reply...

exelban, 2022-12-02:
Please read the readme.

georgedimac, 2022-12-03:
ok, sorry i haven't seen it. also for m1 you can use IOKIT instead of IOHID.

Chamiu, 2023-11-15:
Behave interestingly in this environment.
2.9.11 on macOS Sonoma.1.1 on MacBook Air Late 2020.

Sensor > Module
- Show unknown sensors : OFF -> ON
- HID Sensor : ON -> OFF

If you repeat this operation several times, a list of decent sensors will be displayed.