exelban/stats #1204: Autostart Option disable itself after each restart

DanielOberlechner, 2022-12-03:
Hi,
I'm not able to let the program autostart. I always enable in options the checkbox for autostart,
but after the next reboot its off and I have to start it manually.

Using macOS Ventura Apple M1 Max 16" Macbook.

If more Infos are needed just text me here ...

Kind regards,
Daniel Oberlechner

kambala-decapitator, 2022-12-06:
maybe related: on initial app configuration I selected to not start it at login, but in settings the respective checkbox is ticked

exelban, 2022-12-06:
Thanks, but it's not related. Looks like it's a change in the macOS. It does not support the old way of autostart. I need to research that.

DanielOberlechner, 2022-12-06:
If I click auto login and then pause and start the program via the button the checkbox unchecks itself ...
Does this help further?

DanielOberlechner, 2022-12-06:
You were right, sorry for my dumb question I had it there deactivated and I'm not sooo familiar how macOS works and thought therefore this should do nothing.

Kind regards and thanks for you help,
Daniel Oberlechner