exelban/stats #1205: Triple request to install some helper on start every time after reboot

tkachukcha, 2022-12-04:
**Describe the bug**
Each time I want to manually start fans spinning, after every reboot Stats prompts for installing some unnamed helper. And it does 3 times in a row (have to enter admin password each time). It's very annoying as I have to do it every single day (or more, if I reboot for some reason)
 
**Details:**
 - Device: MacBook Pro 13 2020 M1
 - macOS: 13.0.1
 - Application version: 2.7.46

Jerry23011, 2022-12-04:
Hi, could you please try to
1. Install the [latest version](https://github.com/exelban/stats/releases/tag/v2.8.0)
2. Refer to #1195 and see if it fixes for you

LostInCompilation, 2022-12-08:
Can't confirm. It only asked me once after updating to the new version. Since then, never got asked again when switching to manual or rebooting.

tkachukcha, 2022-12-09:
Updated to 2.8.0, enabled helper in launchctl print-disabled system, rebooted - and still the same. Three prompts asking for password upon manual fan turn on

LostInCompilation, 2022-12-09:
Could you try creating a new user user (admin privileges) for testing? Maybe your user account has some privilege/owner issues.

exelban, 2022-12-09:
@tkachukcha after you enter the password could you control the fans?

tkachukcha, 2022-12-09:
@exelban Yes, I can. 
By the way, also sometimes, in order to start fans again I have to switch from auto to manual and back few times. This issue was before password prompts

@LostInCompilation Will try to

exelban, 2022-12-09:
Please do the next steps if you can:
- run the Stats and install the helper (change the fan speed)
- enter `ls /Library/PrivilegedHelperTools` and check if there are an `eu.exelban.Stats.SMC.Helper`
- after than reboot the mac and run the Stats again. And now try to change the fan speed, but do not give the access
- recheck if `ls /Library/PrivilegedHelperTools` contains the `eu.exelban.Stats.SMC.Helper`

This step allows knowing what is happening. Maybe Stats cannot detect that the helper is already installed.

tkachukcha, 2022-12-16:
Okay, so I restarted, seen that app has been updated to 2.8.1 and now no helper prompts, but also now I can't manually control fans. 
eu.exelban.Stats.SMC.Helper is in /Library/PrivilegedHelperTools

exelban, 2022-12-22:
Could you please check if there is an `eu.exelban.Stats.SMC.Helper` => enabled in the output of the `launchctl print-disabled system`?

tkachukcha, 2022-12-22:
It is enabled, yes

exelban, 2022-12-22:
Please run this build and try to manipulate the fans. In the Documents folder you will find a file log.txt. Please share this file after a run.

[Stats.dmg.zip](https://github.com/exelban/stats/files/10289063/Stats.dmg.zip)

closed 2023-01-09 by exelban