exelban/stats #1206: Toggle the module sign appears on Dashbaord and Settings

Jerry23011, 2022-12-04:
Really ❤️ the app redesign 👍🏻

**Describe the issue**
1. `Toggle the module` indication appears both on **Dashboard** and **Settings**, which might cause confusion
2. `Toggle the module` and `Settings` doesn't seem to be localizable yet
3. Under dark mode, the customization toolbar doesn't have the frame like the one under light mode.

**Screenshots**
> No 1 & 2
[image]
[image]

> No 3
[image]

**Details:**
 - Device: Macbook Pro 14'
 - macOS: 12.6.1
 - Application version: 2.8.0

exelban, 2022-12-04:
Please reinstall the app manually.

exelban, 2022-12-04:
it will fix that issue. `Settings` will be added to the translation. And in the last I don't any issue.

exelban, 2022-12-04:
More interesting it's a dashboard view. Because looks like there is no info in your case. Could you please recheck if it's the same in the light mode too.

closed 2022-12-04 by exelban

Jerry23011, 2022-12-04:
There's a color difference around the widgets under light mode but no color difference under dark mode, I wonder if it is intended that way?
[image]
[image]

exelban, 2022-12-04:
Looks like it's a macOS difference. Because there must be a background for the widget selector.
For example macOS 13:

[image]

exelban, 2022-12-04:
Also redesign is in the progress. There will be more changes. So I don't spend too much time for a clear view. Because it will be changed in the next few weeks.

Jerry23011, 2022-12-04:
I see, thanks