exelban/stats #1211: [Feature request] (Network) Widget Activation Threshold: OR logic instead of AND

dotdash-ben, 2022-12-07:
Hi,

A suggestion as to how this widget would make more sense to me.

**Current situation**: If I put the threshold at e.g. 50 KB, it will only show values if both upload and download exceed this minimum. (AND logic)

**Requested behaviour**: Option to display network traffic when either upload or download exceeds the threshold minimum, or have an option to decouple the two (i.e. show upload from x KB/s and download from y KB/s). (OR logic)

**Reasoning**: I often have a higher download rate which I miss because my upload is so low. I'd still like to be able to see this, without being bothered by the low speeds always being displayed.

exelban, 2022-12-07:
Hi. Thanks for the feedback.
I'm not sure if it's AND now. Right now It's selecting the minimum value between upload and download. And compare it with the threshold. So it's more like OR right now.

dotdash-ben, 2022-12-09:
I don't agree, but I think I didn't phrase it clearly, sorry about that.
I'll try again:

Current: Threshold = 50 KB/s (T), it will look at the minimum value, so if Upload = 20 KB/s (U) and Download = 100 KB/s (D), it won't show anything.

- In case of OR logic (U>T|D>T) it **would** show the values as one of the two values meets the threshold. 
- In case of AND logic (U>T&D>T) it **would not** show the values, as at least one of the values does not meet the threshold. This is the current behaviour, hence my reasoning it's AND logic.

I hope I was able to explain myself better.

And thanks for considering this feature :)

exelban, 2022-12-09:
Ok. I think I catch that.

Try this build: 
[Stats.dmg.zip](https://github.com/exelban/stats/files/10198023/Stats.dmg.zip)

closed 2022-12-18 by exelban