exelban/stats #1218: Total System Power consumption value cut off in Menu bar

LostInCompilation, 2022-12-11:
**Describe the bug**
Value is cut off with 2, 3 or 4 values displayed when static width is enabled. See images.

**Details:**
 - Device: M1 Pro MacBook Pro
 - macOS: 13.0.1
 - Application version: 2.8.0

[image]

[image]

Jerry23011, 2022-12-11:
Hi, you can turn off `Static width` in Widget Settings in the Sensors module, it can temporarily fix the problem.
[image]

LostInCompilation, 2022-12-11:
Yes I know, the issue is only present with static width enabled. I have it disabled most of the time, but tested some things yesterday and saw the bug.

closed 2022-12-18 by exelban