exelban/stats #1224: Sensors module (popup) is missing from the translation files

zigapovhe, 2022-12-12:
Text from sensors module is not in the translation files (check screenshot below). Could you add them please? Thanks! Best of luck!

**Details:**
 - Device: Macbook Pro M1 2021
 - macOS: 13.0.1
 - Stats version: 2.8.1

[Screenshot](https://i.imgur.com/L7IRjlr.png)

closed 2022-12-18 by exelban

zigapovhe, 2023-02-04:
@exelban This is still not fixed. Most of the strings in sensors module is not in translation files. For example if I change language to Chinese (to make it more obvious). All of those English values cannot be translated currently. 

[image]

exelban, 2023-02-04:
there is no translation for the sensors. And for now, there is no plan to do that.