exelban/stats #1225: Incoming connections from 127.0.0.1 all the time. Why?

pictosun, 2022-12-12:
**Describe the bug**
Incoming connections from 127.0.0.1 all the time. Why?

**Details:**
- Device: MacBook Pro M1 Max
- macOS: Ventura 13.0.1
- Application version: 2.8.1

Hi. Comparing Stats to iStat Menus I do get Incoming connections from 127.0.0.1 all the time. iStat Menus does not show this behaviour. Why does it happen with Stats and what is Stats doing there?

Would be nice to get an explanation. 

Checked those connections with Little Snitch.

Thx.

exelban, 2022-12-13:
What do you mean by incoming connection?

pictosun, 2022-12-13:
Thx for your feedback. I found out what's the point.

When ICMP Host-Check is activated (by default) it's showing this behavior as it's constantly doing this requests.

[image]

pictosun, 2022-12-13:
What does this function do in general? 

If I disable it, it works like before. Cannot see any difference.

exelban, 2022-12-13:
It's an internet connection status.

pictosun, 2022-12-14:
Just found out, that even when empty field those 127.0.0.1 are coming. Thought they would be disable it.

And maybe you can also find another solution for those network check. Within iStat menus they do react to some data I guess (there you don't have pings from 127.0.0.1 all the time).

pictosun, 2022-12-14:
Hi. It's not, that I do use 127.0.0.1 as ping.... I'm not talking about outgoing connections.

Talking about connections from my network 'to' stats and that macOS system is showing it as incoming connection. Try it with Little Snitch (Mini) and you can see what I mean.

It's about that my Little Snitch is showing Incoming connections from my whole network 'to' stats. So it's something different.

AlecKinnear, 2026-11-06:
It's strange what Serhiy Mytrovtsiy is doing. There are incoming connections from 104.18.95.41. I submitted a detailed issue report and he simply deleted it without explanation.

The claim is that my issue a duplicate. I can't find the duplicate. This issue is the closest one.

Here is a screenshot of the incoming connection including the IP address.

[image]

I would suggest that anyone using Stats be very concerned about security, with unexplained incoming connections. I'm only running the GPU module and never had Remote enabled or have an account with Serhiy Mytrovtsiy. 

Serhiy, could you please explain why you maintain an open incoming connection to your users computers?