exelban/stats #1229: Network - Auto Not Working

Ashkaan, 2022-12-17:
**Describe the bug**
The autodetection choice of Network Interface does not pick the top connection in the system.  It seems random.  It should choose the top connection based on "Set Service Order".

**Details:**
 - Device: MacBook Pro 14" (M1 Pro)
 - macOS: 13.1
 - Application version: 2.8.1

exelban, 2022-12-17:
The system automatically chooses that. Stats have no way to manipulate that. You can do that if you want to select a specific interface.

closed 2022-12-17 by exelban

Ashkaan, 2022-12-17:
Sorry, I wasn't clear.  I mean the function is not working at all.  This is set to auto and the only interface connection is on WiFi.
[image]

exelban, 2022-12-17:
Could you change the interface manually?

Ashkaan, 2022-12-18:
Yes