exelban/stats #1233: [UI Glitch] Version 2.8.1 Settings window: horizontal & vertical line and left-menus displaced on Catalina

mackonsti, 2022-12-17:
**Describe the bug**
This is a visual bug found in ** Settings** of version 2.8.1 when running Stats on Catalina 10.15.7.

**Details:**
 - Device: MacMini 7,1 (Hackintosh)
 - macOS: 10.15.7
 - Application version: Official release 2.8.1 on GitHub

Apologies for not posting this earlier, I updated Stats on my older hack at my parent's house.
This is a visual glitch on Catalina (including any latest available updates on 10.15.7).
The OS itself has been running perfectly without hiccups or issues, so this could be a UI glitch due to Catalina's displaying capabilities vs. BigSur or Monterey etc. new visual styles.

Previously, Stats had no issue on the ** Settings** window. Attaching a screenshot.

Thank you again for your work and efforts @exelban to you and all the contributors.
Season greetings, all the best.

[image]

exelban, 2022-12-17:
Hi. Thanks for the feedback.
Unfortunately, I have no more mac with 10.15 and earlier. So I cannot test the app on that. I'm making a lot of changes in the design of the setting. And I cannot ensure that everything will work fine on the oldest version of mac. 
I will try to run mac 10.15 to debug these problems. Otherwise, I will drop support macs up to macOS 11. Since I cannot even install the previous macOS because of M1.

exelban, 2022-12-17:
quick hack but still. Could you please try this build?

[Stats.dmg.zip](https://github.com/exelban/stats/files/10252917/Stats.dmg.zip)

mackonsti, 2022-12-18:
Hi @exelban thanks for your help.

Catalina can be installed via some virtual machine, I am sure, if you have the time and energy to spend on it.

Nevertheless, the horizontal line glitch seems fixed, however the vertical side menu on the left is still **displaced** (I still find this weird for retro-compatible apps to have this glitch). Can you think of anything for the side menu?

Also, the container window like in Sensors, scrolls fine from top to bottom without UI glitches. Only that side-thing remains...

Thank you.
[image]

exelban, 2022-12-18:
Thanks.
It's a problem with the macOS. The same views are rendered in totally different ways. Even if macOS 10.15 support the sidebar, it does not work properly. Only starting from macOS 11 it works fine (spacing at least).

I have tried to install Catalina on m1 with parallels. But without success(

Could you try this build:
[Stats.dmg.zip](https://github.com/exelban/stats/files/10253637/Stats.dmg.zip)

mackonsti, 2022-12-18:
Oh, on Apple Silicon there's no way to install x64 systems... apologies for my proposing it, I thought you had an Intel Mac; I should have clarified it.

In any case, **your second fix did the job**, thank you for taking time to have a look, and congrats for this great blind-fix 🤣 

You can officially merge the fix and hopefully 2.8.2 will be released officially soon 😄 

Hopefully, this UI glitch won't become an issue again in future releases of Stats !

One last question, if you allow me (and I will create a different ticket) if FANS are disabled, are they supposed to still show up in the vertical menu drop-down too? Stats is correctly detecting my (only) fan on my Intel NUC, but the others are wrongly reported by SMC (again, it's a Hackintosh) so how can one remove the mention of the "fake" fans in the drop down lists?  Let me know if I can create a new ticket.

Anyway thank you, and season greetings again.
[image]
[image]
[image]

exelban, 2022-12-18:
You must remove them from the fakesmc or whatever you use.

closed 2022-12-18 by exelban