exelban/stats #1244: add option to show cpu throttle

augustobob, 2022-12-24:
If possible, add a option to indicate If the system is throttling the cpu, maybe changing the icon color or a signal.
Macbooks thermals are poor and it gets throttled all the time, I never know if its throttled or not powerful enough.

thanks for considering

LostInCompilation, 2022-12-24:
Would ne a nice feature. Maybe also adding the current CPU clock (GHz) as well. Hope this is also possible with M1/M2 chips.

BitesPotatoBacks, 2022-12-24:
@LostInCompilation 

> Maybe also adding the current CPU clock (GHz) as well. 

This metric is provided in the current version of Stats, but only for Intel 😔.

> Hope this is also possible with M1/M2 chips.

Support for frequency metrics on Apple Silicon would have to use `powermetrics` (which requires root), or Stats could use the counter values within the IOReport like I have in [my project](https://github.com/BitesPotatoBacks/SocPowerBuddy/stargazers) (but the access would have to be done using a XPCService, for IOReport sampling won't work without it within desktop apps)

exelban, 2023-01-09:
Hi. I'm sorry, but it's not possible to do that. @BitesPotatoBacks this way does not work, I have tried that way many times and on different macs. The frequency obtained in this way is too scattered. I don't find any way to obtain a CPU frequency for the apple silicon.

closed 2023-01-09 by exelban

LostInCompilation, 2023-01-09:
@exelban Did you try the code from @BitesPotatoBacks? It works very reliable for me, tested it on two different machines.
I think this would be a good solution.