exelban/stats #1250: "Stats.app" may be damaged or incomplete

silasg, 2022-12-31:
**Describe the bug**
I have Stats running all day long. Every few days to weeks it disappears from my menu bar. In such cases if I try to reopen it, Mac OS tells me

> You can’t open the application “Stats.app” because it may be damaged or incomplete.

`brew reinstall stats` then fixes the issue ... at least for a few days to weeks.

I encounter the issue for a longer time now with different Stats and Mac OS versions. However, all on the same device. I never encountered similar behavior with other apps or some kind of data corruption or similar.

**Details:**
 - Device: Macbook Pro 2019 16"
 - macOS: 13.1 (22C65)
 - Application version: 2.8.0

closed 2023-01-02 by exelban

silasg, 2023-01-02:
"not planned"? Is the reported problem known and a fix is not planned at all? If so, could you please point me somewhere where I could learn more about the causes for it?

I would have expected an answer a la "this is hard to impossible to reproduce" but the "not planned" is kinda unexpected.

Alcas1, 2023-04-02:
Agreed, this is literally breaking my app every week or so. Most people I know just uninstalled it after it broke the first time.

victor-marino, 2024-03-19:
I'm experiencing the same issue. Every few days, especially after a restart or a MacOS update, I get the "Stats.app may be damaged" message, and I can no longer open it.

The only solution is to reinstall it every time, which is obviously a chore.

I think this ticket should be reopened.

I'm on a MacBook Pro 13" 2020 M1, currently running Sonoma 14.4 (but the issue was also present in Ventura).