exelban/stats #1251: Possible to reduce network speed widget width?

dellagd, 2022-12-31:
**Describe the bug**
Is there any way to reduce the width of the network speed widget? My menu bar has a lot of stuff in it already, and it takes up a lot of space, which is usually blank.

If not currently, I seems like speeds could either be single digit, double digit, or stepped up to the next unit and displayed as 0.X unit/s. Doing this looks like it would reduce the overall width a significant bit.

**Details:**
 - Device: Macbook Pro M1
 - macOS: 13.1
 - Application version: 2.8.2

[image]

vuanhson, 2023-01-02:
I think application cannot control this size, iStat Menus is same, it can be wider but cannot be shorter. Need someone confirm this.

BitesPotatoBacks, 2023-01-02:
That extra space seems to be reserved for the pictogram, which (at least for me) was set to "none" by default. The app could definitely make the widget less wide when the pictogram is hidden, maybe also adapting based on the length of the up/down speeds.

closed 2023-01-02 by exelban

andreaswilli, 2023-04-15:
I'd also be interested in this. Hiding the pictogram reduces the width by a little bit, but there is still a lot of whitespace.
Any reason we can't display the numbers like @dellagd proposed and reduce the width?