exelban/stats #1254: Battery: percentage inside icon makes icon red after ~%20

alexander-potemkin, 2023-01-03:
Thank you for a wonderful product!

I discovered I could place a battery icon with percentage inside of it, but found an unexpected side effect: after ~20%, which is plenty of power for M* based laptops, my icon becomes red and there is no way to disable it, keeping percentage inside of the battery.

Is there any way color highlight would be optional?
[image]

alexander-potemkin, 2023-01-03:
btw, colorize is disabled for the battery widget, so it might very well be a bug, I guess?

leocaseiro, 2023-01-05:
You can set the **Low level notification** to **Disabled**

[image]

alexander-potemkin, 2023-01-05:
@leocaseiro , thank you. I don't mind notifications and I actually use them. But it's on 5% for me.

I would like to make sure the icon is not red, as it distracts the attention quite a lot.

[image]

exelban, 2023-01-09:
Hi. Thanks for the feedback.
Yeap, it does not look great. I will find a way how this case could be covered.

alexander-potemkin, 2023-01-09:
@exelban , thank you, appreciate it.
And thanks for the app, I just love it!

closed 2023-02-05 by exelban

alexander-potemkin, 2023-02-06:
@exelban , thanks! Looks good now!! (hope/guess it would remain monochrome when below 20% as well)

[image]

One more question, if you don't mind? What does 'low power mode' switch means?

[image]

exelban, 2023-02-06:
Enabling this option will colorize (to orange) the battery if the `low power mode` is enabled in the macos.

alexander-potemkin, 2023-02-06:
Thanks a lot!! I really appreciate all the work you are doing!

alexander-potemkin, 2023-02-06:
apologies, @exelban , I believe I've found a sub-issue here: it seems like the battery filling doesn't decrease as the battery deplete - please, see screenshot attached

[image]

Despite the fact that the battery is ~30% empty, the background is still 100% full.

Is it possible to change that? If so, would you like some other details or a new issue for that or anything else? :)

exelban, 2023-02-07:
thanks, I will check that.

exelban, 2023-02-08:
Hi. Everything works fine. Just in dark mode there is no big contrast between colors.

alexander-potemkin, 2023-02-08:
That's right, the contrast is not so visible. Would it be too much to ask to change the contrast? Or if I don't like it, I can just stick to the option with no numbers on it?

exelban, 2023-02-08:
Yeap, I will try to make the diff more visible.