exelban/stats #1255: app shows huge amount of RAM, but can't find in activity monitor

flowen, 2023-01-05:
**Describe the bug**
[image]

[image]

Wondering if this is a bug or I can find the process elsewhere. 
Wish I could close the process from stats.app. There's an x when hovering, but nothing happens when clicked.

**Details:**
 - Device: Macbook Pro 2019
 - macOS: 13.1
 - Application version: 2.8.3

flowen, 2023-01-05:
update: I uninstalled the app, and rebooted. 

It's still showing me that 80%+ is in use. Wondering if this is a bug or if I have some app slurping up all my memory
[image]

exelban, 2023-01-09:
Hi. Thanks for the feedback. 
After closing the app from the popup you need to unpin the process by pressing on the row. 

Please share the output of the `top -l 1 -o mem -n 8 -stats pid,command,mem ` next time when you see some strange value in the top processes (like 85GB of usage).

Don't think it's a bug. The system reports these values.

flowen, 2023-01-09:
@exelban thanks for the reply!

I removed the app, but the total RAM usage is similar after a reboot.. and if it was really using 88GB than wouldn't usage be 100%+ ?

thanks for that command, will try next time

exelban, 2023-01-09:
88GB it's a bug in the parsing output from `top`.
Does the ram still stay on the 80% level after you restart the mac?

flowen, 2023-01-09:
yeap

exelban, 2023-01-09:
Strange, it just means that you have a lot of apps on autostart that load your mac. It's not normal to have >50% after the restart. But also it depends on the number of apps that running at that time.

flowen, 2023-01-09:
ok lol yea I had "reopen windows on reboot" checkmark on so it opened the same apps. After rebooting cleanly, it was on 23%. Anyhow, the app, showed 88gb, I will reinstall later and let you know if it happens again

exelban, 2023-01-09:
If some process shows 88GB please share the output of the command above. It's for 100% bug in the parsing output from the `top`.

exelban, 2023-01-15:
Hi. Please post the results of the `top` when you see the process with unnormal ram usage again. I will reopen the issue.

closed 2023-01-15 by exelban