exelban/stats #1258: Fan Control is not working

FrankyWeber, 2023-01-10:
Fan control is not working to control the fan speed manually though the app.
I'm currently running macOS Ventura 13.1.0 on a M1 Max MPB. Stats version is 2.8.3.
Do you know if there is something that can be done?

Appreciate it. Regards,

Franky

exelban, 2023-01-12:
Hi. Thanks for the feedback.
Please change the fan speed in the Stats, and after run `/Applications/Stats.app/Contents/Resources/smc fans`. Please share the output of this command.

FrankyWeber, 2023-01-12:
Man, I don't know what to say. Now it worked, but I would say it is probably working intermittently because it was not working for a few days. I didn't restart my Mac, nor the stats app since I posted here. I'll let you know if it stops again.

```
franky@faust-MBP ~ % uptime
12:40  up 6 days,  3:42, 2 users, load averages: 2.58 3.70 3.92

franky@faust-MBP ~ % /Applications/Stats.app/Contents/Resources/smc fans
Number of fans: 2.0

0: Fan #0
Actual speed: 4486.09326171875
Minimal speed: 1499.0
Maximum speed: 5348.0
Target speed: 4502.0
Mode: forced

1: Fan #1
Actual speed: 4806.2158203125
Minimal speed: 1499.0
Maximum speed: 5776.0
Target speed: 4812.0
Mode: forced

```

FrankyWeber, 2023-01-20:
It just stopped working again and now I could capture the details you asked for when the issue occurs:

[image]

exelban, 2023-01-20:
Hi. Thanks for the info.

please try to change the fan speed with the slider and check again the `smc fans` output.
I'm interested if the `Target speed` is changing when you change the slider value.

FrankyWeber, 2023-01-20:
Each output of the smc fans command was after I changed the speed. The target speed does not change it stays at -1.0

exelban, 2023-01-20:
hmm, does the restart of Stats change anything?
I can make a debug build to collect the logs. Because I have no idea why it does not change the speed(
Give me know if it's ok for you. The logs will be collected locally only.

Another idea. As I see you have enabled fan sync control. Disable it and try to control then.

FrankyWeber, 2023-01-20:
I have tried restarting and also already tried disabling fan sync control, but it doesn't help. I just tried both things again and still does not work.
Well, you do this for free, so it is up to you if you want to debug this issue, I'm fine with running the debug.

exelban, 2023-01-20:
thanks. I will make a debug build later this week.

rafenden, 2023-01-22:
I can confirm the issue also exists on i9 MacBooks.

FrankyWeber, 2023-01-23:
I didn't reboot, I didn't close the stats app and did nothing other than trying to change the speed slider.

[image]

DavidInojosa, 2023-01-26:
Not working here too... Macbook Pro M1. I did restart the mac and close/open stats app, but nothing changes when I slide fans speed.
[image]

exelban, 2023-01-28:
Hi.
Someone who cannot change the fan speed please run this build. And try to manipulate the fans via this build.
After you will find a file `log.txt` inside the Documents folder. Please share that file.

 
[Stats.dmg.zip](https://github.com/exelban/stats/files/10527588/Stats.dmg.zip)

DavidInojosa, 2023-01-30:
> Hi. Someone who cannot change the fan speed please run this build. And try to manipulate the fans via this build. After you will find a file `log.txt` inside the Documents folder. Please share that file.
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/10527588/Stats.dmg.zip)

[log.txt](https://github.com/exelban/stats/files/10533021/log.txt)

apparently the file is empty after run this build(2.8.6) and manipulate fans.

SteelCrow, 2023-01-30:
> Hi. Someone who cannot change the fan speed please run this build. And try to manipulate the fans via this build. After you will find a file `log.txt` inside the Documents folder. Please share that file.
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/10527588/Stats.dmg.zip)

Same story with fan control for me.

- I tried to build it from the master branch, but nothing changed.
- I tried your version and it worked the same way.

[log.txt](https://github.com/exelban/stats/files/10534065/log.txt)

exelban, 2023-01-30:
Do not try the release, but the build I shared. Only that build will write logs to the file.

Hmm, I need to add more breakpoints.

IwVr, 2023-02-02:
[log.txt](https://github.com/exelban/stats/files/10569325/log.txt)

[image]
Fans are still not changing. Haven't changed anything recently. I think the issue might have popped up when I tried to use the manual sliders for the first time. Usually, I just use the max setting.

> Do not try the release, but the build I shared. Only that build will write logs to the file.
> 
> Hmm, I need to add more breakpoints.

exelban, 2023-02-02:
@IwVr according to the logs (and also a screenshot) your fans are in automatic mode. And it wasn't changed. It's strange.

IwVr, 2023-02-14:
@exelban Any Updates? If not let me know, I'll have a go at figuring out the issue.

exelban, 2023-02-19:
Hi. So, first of all, we need to figure out if your fans could be controlled. It's strange that from time to time it works.

There are commands that allow to manually change the fans speed:
Change the fan mode: 
`sudo /Applications/Stats.app/Contents/Resources/smc fan 0 -m 0` - automatic mode
`sudo /Applications/Stats.app/Contents/Resources/smc fan 0 -m 1` - manual mode

Only in the manual mode, the fan speed could be changed. The fan speed could be changed with:
`sudo /Applications/Stats.app/Contents/Resources/smc fan 0 -v 2000`

to see the actual status:
`/Applications/Stats.app/Contents/Resources/smc fans`

Please try to change the mode and after the fan speed. Give me know if on some step you will have some problems. For example, if after changing the mode, the fan mode does not change, etc.

FrankyWeber, 2023-02-20:
Hi,

I changed it using the commands you provided and the fans are spinning in manual/forced mode:

```
Last login: Sat Feb 18 11:10:58 on ttys000
franky@faust-MBP ~ % /Applications/Stats.app/Contents/Resources/smc fans
Number of fans: 2.0

0: Fan #0
Actual speed: -1.0
Minimal speed: 1499.0
Maximum speed: 5348.0
Target speed: -1.0
Mode: automatic

1: Fan #1
Actual speed: -1.0
Minimal speed: 1499.0
Maximum speed: 5776.0
Target speed: -1.0
Mode: automatic

franky@faust-MBP ~ % sudo /Applications/Stats.app/Contents/Resources/smc fan 0 -m 1
Password:
franky@faust-MBP ~ % /Applications/Stats.app/Contents/Resources/smc fans
Number of fans: 2.0

0: Fan #0
Actual speed: -1.0
Minimal speed: 1499.0
Maximum speed: 5348.0
Target speed: -1.0
Mode: forced

1: Fan #1
Actual speed: -1.0
Minimal speed: 1499.0
Maximum speed: 5776.0
Target speed: -1.0
Mode: automatic

franky@faust-MBP ~ % sudo /Applications/Stats.app/Contents/Resources/smc fan 1 -m 1
franky@faust-MBP ~ % /Applications/Stats.app/Contents/Resources/smc fans
Number of fans: 2.0

0: Fan #0
Actual speed: -1.0
Minimal speed: 1499.0
Maximum speed: 5348.0
Target speed: -1.0
Mode: forced

1: Fan #1
Actual speed: -1.0
Minimal speed: 1499.0
Maximum speed: 5776.0
Target speed: -1.0
Mode: forced

franky@faust-MBP ~ % sudo /Applications/Stats.app/Contents/Resources/smc fan 0 -v 2000

franky@faust-MBP ~ %
franky@faust-MBP ~ %
franky@faust-MBP ~ % sudo /Applications/Stats.app/Contents/Resources/smc fan 0 -v 2000
franky@faust-MBP ~ %
franky@faust-MBP ~ % sudo /Applications/Stats.app/Contents/Resources/smc fan 1 -v 2000
franky@faust-MBP ~ % /Applications/Stats.app/Contents/Resources/smc fans
Number of fans: 2.0

0: Fan #0
Actual speed: 1992.285400390625
Minimal speed: 1499.0
Maximum speed: 5348.0
Target speed: 2000.0
Mode: forced

1: Fan #1
Actual speed: 467.275146484375
Minimal speed: 1499.0
Maximum speed: 5776.0
Target speed: 2000.0
Mode: forced

franky@faust-MBP ~ %
franky@faust-MBP ~ % sudo /Applications/Stats.app/Contents/Resources/smc fan 1 -v 3000
franky@faust-MBP ~ % sudo /Applications/Stats.app/Contents/Resources/smc fan 0 -v 3000
franky@faust-MBP ~ % /Applications/Stats.app/Contents/Resources/smc fans
Number of fans: 2.0

0: Fan #0
Actual speed: 3016.742919921875
Minimal speed: 1499.0
Maximum speed: 5348.0
Target speed: 3000.0
Mode: forced

1: Fan #1
Actual speed: 3019.576904296875
Minimal speed: 1499.0
Maximum speed: 5776.0
Target speed: 3000.0
Mode: forced

franky@faust-MBP ~ % sudo /Applications/Stats.app/Contents/Resources/smc fan 0 -v 3200
franky@faust-MBP ~ % sudo /Applications/Stats.app/Contents/Resources/smc fan 1 -v 3200
franky@faust-MBP ~ % /Applications/Stats.app/Contents/Resources/smc fans
Number of fans: 2.0

0: Fan #0
Actual speed: 3216.784423828125
Minimal speed: 1499.0
Maximum speed: 5348.0
Target speed: 3200.0
Mode: forced

1: Fan #1
Actual speed: 3211.017333984375
Minimal speed: 1499.0
Maximum speed: 5776.0
Target speed: 3200.0
Mode: forced

```

exelban, 2023-02-20:
@FrankyWeber hi. Thanks. And it's not possible to control fans via Stats at the same time?

FrankyWeber, 2023-02-20:
I'm not running your debug build. Still running the previous one that was not working and just confirmed... when I try to change the fans speed from the stats.app itself it does not do anything.
[image]

exelban, 2023-02-20:
please try to switch between automatic and manual mode and check if it's changing in the terminal.

FrankyWeber, 2023-02-20:
By using the app nothing happens. I switched to auto via CLI and changed to manual via the app, nothing happened. Then switched to manual via CLI then it worked. Tried to change the speed via the app and it didn't work, then manually via CLI it worked. Tried to change back to auto from the app, nothing happened, then I changed via CLI.

exelban, 2023-02-20:
Do you have `eu.exelban.Stats.SMC.Helper` in the `ls /Library/PrivilegedHelperTools`?

FrankyWeber, 2023-02-20:
I do have it:
franky@faust-MBP ~ % ls -lh /Library/PrivilegedHelperTools/eu.exelban.Stats.SMC.Helper
-r-xr--r--  1 root  wheel   189K Jan 21 21:24 /Library/PrivilegedHelperTools/eu.exelban.Stats.SMC.Helper

exelban, 2023-02-20:
please run this build, try to manipulate the fans with UI, and share the `log.txt` that could be found in the `Documents` folder.
Please change a few times from auto to manual mode from UI to gather the logs about the mode change functionality.

[Stats.dmg.zip](https://github.com/exelban/stats/files/10786813/Stats.dmg.zip)

IwVr, 2023-02-28:
> please run this build, try to manipulate the fans with UI, and share the `log.txt` that could be found in the `Documents` folder. Please change a few times from auto to manual mode from UI to gather [...]

- I've ran the attached build, still not working, tried multiple ways
- Attached is the log.txt
- Also below is a screenshot of the terminal showing fan status

Let me know next steps

[log.txt](https://github.com/exelban/stats/files/10849862/log.txt)
[image]

exelban, 2023-03-01:
Thanks, please try this build and share the output. There is no changes of how it works, but there are more logs added that allows to debug the problem.

[Stats.dmg.zip](https://github.com/exelban/stats/files/10860426/Stats.dmg.zip)

sanderDijkxhoorn, 2023-03-09:
> Thanks, please try this build and share the output. There is no changes of how it works, but there are more logs added that allows to debug the problem.
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/10860426/Stats.dmg.zip)

The normal release build & this custom build are not working for me on Ventura M1 Pro
[image]

[image]

exelban, 2023-03-09:
@sanderDijkxhoorn please read all posts related to the debug build.

sanderDijkxhoorn, 2023-03-09:
> @sanderDijkxhoorn please read all posts related to the debug build.

Setting the fan speeds using terminal works fine

SteelCrow, 2023-03-12:
> Removing and installing (reinstalling) the helper fixed the issue, thanks.

The same here.

exelban, 2023-03-12:
Ok, so if some will want to help with fixing that problem just ping me. I'm going to close this issue due to inactivity.

closed 2023-03-12 by exelban

Rakshit-Yadav, 2024-08-30:
I have read the whole thread and I am currently running macOS Sonoma 14.6.1 on M2 Pro laptop.
I have tried it setting through sensor gui and terminal command but it gives this output:

[image]

I Do not get the messages about helper file and removing and reinstalling it and a bit confused regarding that. Can someone who has fixed this issue for themselves can explain the fix here in detail in one post for others to try please.

Thank you in advance