exelban/stats #1259: Disks widget shows removable disk even though it's disabled in settings

LostInCompilation, 2023-01-10:
**Describe the bug**
The disk widget pop up is really small. Would be nice if it could be made larger and look like the network widget.

**Details:**
 - Device: MacBook Pro M1 Pro 2021
 - macOS: 13.1
 - Application version: 2.8.3

## Current look:
[image]

## Would be better if it looks like this:
[image]

exelban, 2023-01-12:
Hi. There are no plans to change the design of the disk view.

closed 2023-01-12 by exelban

LostInCompilation, 2023-01-12:
OK but why not? It looks a bit out of place.

exelban, 2023-01-12:
There is a lot more important thing to do. I don't have any time for the nearest time. And I don't see any problems with the current design.