exelban/stats #1264: Low power mode status always "Disabled"

effleurager, 2023-01-14:
Low power mode does not seem to be detected, whether only on battery or always. This has previously worked, but I'm not certain if it's a change in macOS or Stats that's broken it.

[image]

exelban, 2023-01-15:
Hi. Thanks for the feedback.
I take a look, and there is a subscription for low-power mode [events](https://developer.apple.com/documentation/foundation/nsnotification/name/1617322-nsprocessinfopowerstatedidchange). But looks like it does not work now.

Looks like apple broke something. Because it works previously for sure. And there is no info about deprecation. The only solution it's waiting till apple will fix that.

Ping me after the next macOS update. I will try to find another way for that if apple will not fix that.

closed 2023-01-15 by exelban

Drallas, 2023-01-25:
> Hi. Thanks for the feedback. I take a look, and there is a subscription for low-power mode [...]

@exelban Don't know if Apple is even aware of it, since their battery app show the correct information. I send them Feedback (FB11967793) with the Feedback Assistant. Perhaps it helps.

effleurager, 2023-01-27:
Updated to Ventura 13.2 (22D49), and this works flawlessly now! 🙌🏽

Drallas, 2023-01-27:
Running the same version (Ventura 13.2 (22D49)) here but still no response. Did deleted the stats .plist in preferences and restarted, but it's stuck on enabled now if when Lower Power Mode is reported on in macOS.

iamyeizi, 2023-02-17:
> Running the same version (Ventura 13.2 (22D49)) here but still no response. Did deleted the stats .plist in preferences and restarted, but it's stuck on enabled now if when Lower Power Mode is reported on in macOS.

Something similar happens to me. I'm on v13.2.1 (22D68). When I turn on LPM and it doesn't change that icon to orange until I close and open the app, the same for when I turn off LPM and want that orange fill to go away. IMO it could be something about the cache management of a certain function or the trigger to the system info @exelban 

BTW excellent app ❤️

movinimage, 2023-03-12:
Same issue. On 13.2, M1 Max. Low Power shows as Enabled no matter the actual state.

skylon07, 2023-05-09:
> Something similar happens to me. I'm on v13.2.1 (22D68). When I turn on LPM and it doesn't change that icon to orange until I close and open the app, the same for when I turn off LPM and want that [...]

Yeah, I'm still running Monterey 12.6.5 (I'm a lazy boi sometimes) and I see the *exact* same behavior; when Stats loads, it displays the correct state of LPM, but if I change LPM (in settings, via command line, whatever) then Stats won't reflect the new state unless I restart it completely. Not even turning the widget on/off fixes it.