exelban/stats #1266: Writing to a multiple partition drives shows traffic on all partitions

LostInCompilation, 2023-01-16:
**Describe the bug**
I have an external drive with two partitions. When I write to one partition, Stats is showing traffic on both partitions equally.
I'm writin to the "M.2 Extern Mac" partition in the screenshot, but it gets duplicated to the "M2 Windows" partition as well.

This is the same drive which is responsible for issue #1260. I still think there is an error of how the API is accessed by Stats. I will take a look at the source code for fixing #1260 when I have time.

**Details:**
 - Device: MB Pro 2021
 - macOS: 13.1
 - Application version: 2.8.4

[image]

exelban, 2023-01-28:
Yes. Because the API allows getting IO only for the drive. Not a logic partition.

closed 2023-01-28 by exelban

LostInCompilation, 2023-01-28:
I did some more testing.
It works for logic partitions but something is a bit off:
The drive shown in this issue is a normal M.2 NVMe internal SSD in an M.2 to USB-C enclosure. That drive is shown in stats as "internal" and also both partitions show the same read/write when writing only to one partition.

If I connect an external HDD with the same partition scheme, it gets reported as "external" and also writing to one partition is only showing traffic on that single partition.