exelban/stats #1267: Opening a menu item returns to Space (virtual desktop) it was first openned on

glenvan, 2023-01-16:
**Describe the bug**
Perhaps related to the fix for [issue 1262](https://github.com/exelban/stats/issues/1262), there is a recent behaviour where accessing a menubar item returns me to the Space (virtual desktop) that it was first opened on.

For example:

1. Launch Stats.app
2. Navigate to Desktop 1 using Spaces
3. Open CPU panel from a menubar item
4. Navigate to Desktop 2 or any other Desktop using Spaces
5. Open CPU panel again from a menubar item
   - Note that you will be returned to Desktop 1

On my system I have `Preferences -> Mission Control -> Displays have separate Spaces` enabled and three monitors. Interestingly, the Desktop affinity can be broken by opening the same Stats panel on a different monitor, although that monitor will then exhibit Desktop affinity.

**Details:**
 - Device: Macbook Pro 14-inch, 2021
 - macOS: 12.6.2
 - Application version: 2.8.4

glenvan, 2023-01-16:
Also, let me take this opportunity to thank you all for this fantastic utility and compliment you on the quality of your work. It's just fantastic and I'm grateful that this group donates their time to create and maintain this project.

exelban, 2023-01-16:
Hi. Thanks for the feedback.
Hmm, yeap. I remember now how I fixed that problem with separate spaces. And looks like the fix of #1262 was a solution for that. I will take a look and will try to find a solution for both cases. Thanks.

gabrielrojasc, 2023-01-20:
Is there a date for the release of this fix?

exelban, 2023-01-20:
Yeap, this Sunday.

LostInCompilation, 2023-01-20:
This will break #1262 again, I guess?

LostInCompilation, 2023-01-20:
@exelban Take a look into this: https://developer.apple.com/documentation/appkit/nswindow/collectionbehavior/1419246-canjoinallspaces

Could be a fix for both, #1262 and this issue.

closed 2023-01-22 by exelban