exelban/stats #1268: Clicking on a menu bar item switches desktops

LostInCompilation, 2023-01-17:
**Describe the bug**
Some new weird bug appeared. I'm using multiple desktop views. To reproduce this, quit Stats and reopen it again.
If I then click on for example the CPU item to open it's pane on Desktop View 1, the CPU pane will always open on Desktop 1. If I am currently on Desktop 2 and click on the CPU pane, my desktop switches back to 1 and Stats shows the CPU pane.
What ever item (CPU/RAM/Disk/Network/...) you open the first time after opening the Stats App, it will then always open on the Desktop you first opened it.
Previously that wasn't the case. I could open the Stats panes on any Desktop i want.

Here's a short video of the issue: https://youtu.be/TLUMG6iCDxk

**Details:**
 - Device: MBPro 2021
 - macOS: 13.1
 - Application version: 2.8.4

exelban, 2023-01-17:
Duplicate of #1267

closed 2023-01-17 by exelban

LostInCompilation, 2023-01-17:
Whoops. Didn't reload the Issues page. Had that issue written yesterday but didn't post it yet.