exelban/stats #1271: GPU percentage issue on MacBook Pro M1 Pro

3singes, 2023-01-19:
Hello,

I just installed Stats, it's really a great plugin, thanks !

There seem to be one issue on OS 13.1, the GPU percentage always stays at zero, and the GPU to show isn't available in the settings.

Thanks, kind regards.

 - Device: Macbook Pro 16" - M1 Pro
 - macOS: 13.1
 - Application version: 2.8.4

exelban, 2023-01-19:
Hi. Thanks for the feedback.
Could you please share the screenshot of the GPU popup view?

3singes, 2023-01-19:
Sure, here it is.
[image]

exelban, 2023-01-19:
enable the module

closed 2023-01-19 by exelban

3singes, 2023-01-19:
Wow, I apologize. I must have enabled it by mistake and simply didn't see it... I work too much, I guess... :)