exelban/stats #1275: Cannot bring detached pane to foreground using mission control

LostInCompilation, 2023-01-22:
**Describe the bug**
Issue #1262 had been broken again by fixing #1267 with commit 0c77346 (rollback for #1262).

Possible fix for both issues:
1) https://developer.apple.com/documentation/appkit/nswindow/collectionbehavior/1419188-stationary
2) https://developer.apple.com/documentation/appkit/nswindow/collectionbehavior/1419246-canjoinallspaces

**Details:**
 - macOS: 13.1
 - Application version: 2.8.5

closed 2023-01-23 by exelban

LostInCompilation, 2023-01-23:
Well I just tried it with a simple window. It does solve the problem. The window opens on current space, stays open when switching spaces, and mission control has no effect (it always stay at the top).

exelban, 2023-01-23:
I don't know what you try and what works in your case. But:
- stationary will always open the windows only in the workspace the window was opened first. (if I open CPU popup in the first workspace it cannot be opened in the second, it always will be shown at the first) -> it does not work at all!
- canJoinAllSpaces works fine with multiple workspaces (with a little bit of strange behavior, but still). But it does not fix the mission control problem

So, you are free to make a fork and use some of your proposals if it works for you. But none of these will not be implemented in the Stats.

LostInCompilation, 2023-01-23:
No I meant in conjunction with `moveToActiveSpace`, not only using stationary. Stationary is a workaround for the foreground issue.
I will make some tests tomorrow and let you know 👍🏻

LostInCompilation, 2023-01-23:
Nope I was wrong, there are still issues with my approach.
However `self.collectionBehavior = [.auxiliary, .canJoinAllSpaces]` seems to be working fine, but that requires macOS13.0+ deployment target.

exelban, 2023-01-28:
I have tried this behavior, not covering all cases(