exelban/stats #1281: How does the Fan Helper affect the default behaviour?

nicolopadovan, 2023-01-27:
I recently installed Stats and I love it.
I am currently using a Mac M1 Max, and I was wondering: how does the fan helper affect the default fans behavior?
In particular, I was wondering:

1. Does the fan helper set on "automatic" affect the default fans behaviors? If so, how?
2. Are all the "manual" settings "safe"? Can I crank up the fan speed as I like?
3. Is setting the "Synchronize fan's control" flag to true beneficial? If so, how?

Thank you.

exelban, 2023-01-27:
Fan helper is only a tiny "proxy" application that allows controlling the fans. Because controlling the fans requires root access. And to avoid running the whole Stats app with root, only a small app is running as a root and allows to control the fans.

1. Automatic mode means that the system controls the fans
2. Yes and no. You can change the fan speed via Stats only in the range that macOS reports
3. Not sure understand that. If you want to control all fans with one slider you can do that. Otherwise do not use it.

Nope, but looks like it's a missing feature. It must be added.

nicolopadovan, 2023-01-27:
> 

Thank you for your reply.
I have still some doubts regarding the fan control feature.
As you said, automatic mode means that the system controls the fans: does system, in this case, mean the Stats system or the Apple System? Automatic mode, if I am getting this right, would then mean "Apple default fans behavior".
Also, I can change the fan speed only in the range that macOS reports: is this range the range of the sliders?

Thank you.

exelban, 2023-01-27:
Apple system. Yes.

You can only change the fan's speed in the "safe" range (via Stats). And slider side values are the min and max values that the fan could be run.

closed 2023-01-28 by exelban

gabimoncha, 2025-01-08:
hi @exelban i've noticed that when stats fan control is set on automatic, Apple never increases the RPM based on the CPU / GPU temperature. is this a macOS bug or stats bug?