exelban/stats #1282: feature request: dedicated widgets for sensors

nsklaus, 2023-01-27:
**feature request: dedicated widgets for sensors**
show more information on a particular sensor by having a dedicated widget.

Currently we have for CPU and GPU:  when i click on those in the menu bar, a dedicated widget appears and shows me their usage history through graphs. the idea would be to have the same for a given sensor.
for example: i have CPU Temperature sensor in the menu bar. when i would click on it, it would then show widget displaying a graph with usage history only for that particular sensor. so i could see for example how temperature was very high for some time, but then two minutes ago it started getting back to more acceptable temp.

Right now as it is implemented, when i click on that sensor on the menu bar, i get a rather big general widget that show me all values for all sensors. so, rather than having that, showing dedicated info for that particular sensor only, would be nice. basicaly it's what you've done for cpu and gpu, but replicating that for sensors.

**Details:**
 - Device: macbook pro m2 14" 2022
 - macOS: 13.2
 - Application version: 2.8.5

exelban, 2023-01-27:
If you want to see sensor data you can use the Sensors module for that. There you can see any of the sensors.
There will be no widget per specific sensor.

closed 2023-01-27 by exelban

nsklaus, 2023-01-27:
what about viewing usage history then ? you did seem to find that useful to display for cpu and gpu.
why that would not be desirable as well for sensors ?

LostInCompilation, 2023-01-28:
#1216 is planned (I think). With that you could hide all sensors except for one or two.

nsklaus, 2023-01-28:
@LostInCompilation 
well, yes, that would be the other way around to unclutter the current sensor widget. i thought heat history graph would have been good to have too. but sure, yes, #1216 will be nice to have.