exelban/stats #1284: cpu usage widget shows gpu temp

nsklaus, 2023-01-28:
**Describe the bug**
the cpu usage widget (popup window) shows two graphs about the cpu.
the bigger one shows cpu usage.
the smaller one is supposed to shows cpu temp. but instead it shows gpu temp.
please see attached screenshot below.

**Details:**
 - Device: Macbook Pro M2 14" 
 - macOS: 13.2
 - Application version: 2..8.5

[image]

exelban, 2023-01-28:
looks like apple again makes a mess with sensors in the new silicon.
Could you please share the screenshot of the `sensors` popup view?

nsklaus, 2023-01-28:
now that you ask about another screenshot, the behavior described above is not observable anymore..
computer often behave like that. when you need something it becomes unavailable ;)
i had made another screenshot earlier, for other reasons (i wanted to talk about the difference of values between sensor `power>system total` and `energy>total system consumption`) but it can serve here too as it shows below room temperature values.
and, i've been testing Stats for 3days, it is not the first time i've seen those low values (around 6°C).

here it is:

[image]

nsklaus, 2023-01-28:
and there, those weird values are back, right now, here's a fresh screenshot of it:

[image]

BitesPotatoBacks, 2023-01-28:
@nsklaus All the low values apply to the cpu sensors, so I don't think stats mixed up the cpu widget temp with the gpu temp... but those readings are definitely messed up from Apple changing stuff.

nsklaus, 2023-01-28:
@BitesPotatoBacks yes, you were faster than me, i was updating my post but you replied first :)
you are right. the low values below romm temp are those of the cpu.

BitesPotatoBacks, 2023-01-28:
@nsklaus Haha sorry 😅

nsklaus, 2023-01-28:
so now i wonder if i should close this issue and create another one, or just leave it as it is ?
there's a bug. just, it's not the one i thought it was.
and maybe there's nothing Stats could do about it, if apple reports wrong values and Stats just happen to grab use those. maybe that should be a bug report for apple instead of Stats ?

exelban, 2023-01-29:
Looks like Apple has new temperature sensors in the M2(
It requires debugging. I think will be better to have a new issue with missing the wrong M2 CPU sensors. Then everyone could help who has new silicon.

closed 2023-01-29 by exelban

nsklaus, 2023-01-29:
understood. i create a new issue.