exelban/stats #1285: Total system consumption never stops going up

nsklaus, 2023-01-29:
**Describe the bug**
sensor value for `energy>total system consumption` goes up and up and never seems to stop.
please look at both screenshots below. notice the clock.. in a little less than hour time, it gradualy climbed up from 3.92wh to 9.04wh despite the `average system total` was always around 5.92w (below 6w). by the time it took me to write this issue `energy>total system consumption` has now climbed to 10wh.. (system clock is now 1am, so now 1h and 10 minutes). 
am i misunderstanding something here ? shouldn't it go down a bit if system is left idling around 6w ?

**Details:**
 - Device: macbook pro M2, 14", 2023
 - macOS: 13.2
 - Application version: 2.8.5
 

[image]

[image]

battall, 2023-01-29:
Total System Consumption is how much energy your macbook used since it booted (or after stat resetted), it is in Wh unit (not W), which roughly means energy used, if you take a look at your electricity bill there should be a counter of that too, in KWh probably.

Watt (W) = Volts * Amps = energy usage
Watthour (Wh) = W * Hour = energy used

if you have a 40W light bulb, it will use 40Wh of energy every hour. it is same for macbook.

nsklaus, 2023-01-29:
right, but then if the 40w light bulb suddenly does only 6w, shouldn't the watt-hour value reflect that change and not continue to say "it will use 40wh of energy every hour" when in fact it now does only 6.. the watt-hour should reflect that change and say : that light bulb will now consume 6wh of energy every hour. is it not so ? i could understand if the 40wh value would then decrement slowly because it is an average value. but what i can not understand is when i switch my laptop into low power mode, leave it idling, dim the screen brightness and all, and yet i see the wh value keep on increasing.
as if my laptop will consume more and more power every hour.

nsklaus, 2023-01-29:
"if you have a 40W light bulb, it will use 40Wh of energy every hour. it is same for macbook."
if i use that 40W lightbulb for two hours, that doesn't make it a 80W lightbulb. reporting 80wh for a 40W light bulb used for 2 hours is wrong. value reported should be 40wh the first hour, just as well as the 2nd hour. light bulb doesn't change properties. WH is the measure of consumption over an hour. it doesn't matter if you measure 2hours or 4days. lightbulb will still consume 40watt per hour.

LostInCompilation, 2023-01-29:
Your confusing W (power per time [J/s]) and Wh (Work). Energy (Joule) per hour (J/h=W) is energy usage per time. Wh is used power (total amount of work) after a given time frame (integration of Watts over time). Just simple physics. Integral(J/t dt) = Wt
And of course Stats is correct. It reports the Integral over the Power (=Work). By definition, since the power can never be negative, the integral of the power must always be rising.

Think of it that way:
You have a 40W Lightbulb. It runs for one hour and consumed 40Wh of energy. No you let the bulb run with 10W for another our. In this second hour the energy consumed will be 10Wh. In total the energy consumption in those two hours is 40Wh+10Wh=50Wh.
To make it more clear: A battery's capacity is usually given in mAh (current*time). It can be converted to Wh by factoring in the voltage. Current macs have a 100 Wh battery capacity. If you take 40Wh out of it, 60Wh are left. If you then only take 10Wh out of the battery, the total amount of energy left in the battery will not rise again, it will drop by another 10Wh, leaving 50Wh left.

But anyway, Stats and physics is 100% right. Read about Energy/Work (J, Ws) and Power (W, Energy/time) and whats the difference between them. What you are thinking of is basically "Watt per time (W/s)" which does not exist.

nsklaus, 2023-01-29:
i see that apparently i had a misconception of the thing. thank you for taking time to tell me about it.
i thought it was obvious but apparently there's some sort of subtlety that i missed. i will look into it. 
and you're also right in that case, here, would not be the place to discuss it further, since Stats is complying with the norm about it.

closed 2023-01-29 by nsklaus