exelban/stats #1286: sensors on m2, currently report broken values

nsklaus, 2023-01-29:
**Describe the bug**
The sensors on M2 pro are currently reporting broken values. for example cpu temp is reported as below room temperature. here room is 20°C, and cpu temp sensor says: 6°C
please see attached screenshot down below.

**Details:**
 - Device: macbook pro 2023 (14", M2 pro)
 - macOS: 13.2
 - Application version: 2.8.5
 

[image]

exelban, 2023-01-29:
Please run `/Applications/Stats.app/Contents/Resources/smc list > sensors.txt`.
This command collects all sensor's key/value pairs and saves them to the `sensors.txt`. Could you share that file?

nsklaus, 2023-01-29:
yes, certainly, here it is:

[sensors.txt](https://github.com/exelban/stats/files/10529929/sensors.txt)

exelban, 2023-01-31:
Hi. Could you please share the output of `system_profiler SPHardwareDataType | awk "/Identifier/"`?

nsklaus, 2023-01-31:
yes, here it is:
```
Downloads % system_profiler SPHardwareDataType | awk "/Identifier/"
      Model Identifier: Mac14,9
Downloads % 
```
i think it was shown in the screenshot.

exelban, 2023-01-31:
Thanks. Oh, I missed that(

tomkoen, 2023-02-01:
Tpxx values seem to display correct values, but only under load.
On idle they are ridiculously low, e.g. 5-6 (C)

exelban, 2023-02-01:
The only way to figure out what sensors are responsible for what to run tests. And looks at what sensor changes the value. I don't have any new macs, so I cannot help with that.

looks like some of these could be cores:

```
[TPD0]     25.763214111328125
[TPD1]     25.585601806640625
[TPD2]     25.674407958984375
[TPD3]     25.585601806640625
[TPD4]     25.940811157226562
[TPD5]     25.940811157226562
[TPD6]     25.852005004882812
[TPD7]     25.763214111328125
[TPD8]     25.674407958984375
[TPD9]     25.852005004882812

[TPDa]     25.852005004882812
[TPDb]     25.408004760742188
[TPDc]     25.940811157226562
[TPDd]     25.674407958984375
[TPDe]     25.852005004882812
[TPDf]     25.763214111328125
[TPDg]     26.20721435546875
[TPDh]     25.585601806640625
[TPDi]     25.852005004882812
[TPDj]     25.674407958984375

[TRD0]     26.296005249023438
[TRD1]     26.20721435546875
[TRD2]     26.02960205078125
[TRD3]     26.118408203125
[TRD4]     26.473602294921875
[TRD5]     26.384811401367188
[TRD6]     26.296005249023438
[TRD7]     26.473602294921875
[TRD8]     26.296005249023438
[TRD9]     26.384811401367188

[TRDa]     26.384811401367188
[TRDb]     26.02960205078125
[TRDc]     26.20721435546875
[TRDd]     26.296005249023438
[TRDe]     26.384811401367188
[TRDf]     25.940811157226562
[TRDg]     26.296005249023438
[TRDh]     26.02960205078125
[TRDi]     26.473602294921875
[TRDj]     26.20721435546875
```

JonyEpsilon, 2023-02-01:
At least from casual inspection, the sensors are reporting something sensible most of the time for me (Mac 14,5): run a benchmark and the "CPU * core *" sensors in the menu read appropriately high temperatures, and then cool down when the CPU is idle. It's just that some small-ish fraction of the time they report nonsensical near-freezing-cold values!

exelban, 2023-02-01:
maybe you can check if some values from that list are similar to the CPU sensors that are already in the app when benchmarking the system?

you can show all sensors in the Stats by enabling `Show unknown sensors`

tomkoen, 2023-02-02:
Somehow this app displays it all correctly:
https://www.tunabellysoftware.com/tgpro/releasenotes/
"Improved the CPU temperature reporting for the new M2 Pro/M2 Max models when the usage is low so the UI can continue to update properly"

wongkk25, 2023-02-09:
Also have a M2 Pro machine here (10 cpu/16 gpu cores; model identifier Mac14,9). CPU and GPU temps also looked off on stats v1.8.6, and don't appear at all in v1.8.7 (probably cause of the addition of the enums in `SystemKit.swift` but no m2 pro temp sensors in `values.swift`?).

I'm attaching three `smc list -t` dumps, hope these help narrow down the correct sensor codes:
[sensors-cool.txt](https://github.com/exelban/stats/files/10694462/sensors-cool.txt): temperatures from idle/simple web browsing usage
[sensors-hot.txt](https://github.com/exelban/stats/files/10694463/sensors-hot.txt): temperatures after running multiple instances of `yes > /dev/null` to invoke 100% CPU usage
[sensors-hotgpu.txt](https://github.com/exelban/stats/files/10694468/sensors-hotgpu.txt): temperatures after running GFXBench Metal benchmark to invoke 100% GPU usage

exelban, 2023-02-09:
@wongkk25 hi. Thanks for sharing that data. It really helps.
Yeap, in the latest release there are no m2 pro/max sensors. I don't know the sensors, it is better to disable them now.

But it looks like we are close to the dictionary. Please take a look at these sensors. Maybe someone can confirm my guesses.

If someone has M2 max, please share the sensors list. Maybe it will help with research.

[sensors-hotgpu.txt](https://github.com/exelban/stats/files/10698654/sensors-hotgpu.txt)
[sensors-hot.txt](https://github.com/exelban/stats/files/10698655/sensors-hot.txt)
[sensors-cool.txt](https://github.com/exelban/stats/files/10698656/sensors-cool.txt)

zaubara, 2023-02-09:
Sure, here's the sensors for an MacBook Pro 14" M2 Max in idle:
[sensors.txt](https://github.com/exelban/stats/files/10699455/sensors.txt)

I also had the temps reported low in the version before. Same goes for Mx Power Gadget - they also report the wrong idle values. Weird!

exelban, 2023-02-13:
Hi. Could someone try this build?
[Stats.dmg.zip](https://github.com/exelban/stats/files/10725026/Stats.dmg.zip)

ryneb, 2023-02-13:
That build appears to have temperatures working again for me on an M2 Max: 
[image]

wongkk25, 2023-02-13:
@exelban hi, CPU temps on the new build seem good to me. A few things that caught my attention:
- GPU core temps rise with CPU load despite not running any GPU intensive tasks, but this may be normal with a SoC architecture
- Only 8 GPU cores show up instead of 16 for my machine

If it helps for code validation purposes, I've also attached a list of SMC codes that appear on the m2 max and not on the m2 pro (based on the provided sensor codes in this thread): [diff-codes.txt](https://github.com/exelban/stats/files/10725177/diff-codes.txt)

zaubara, 2023-02-13:
I'm afraid that those are the wrong sensors. The temps should be much higher when running cinebench. Number of CPU cores is correct though, however only 8 GPU cores are shown for me too, their temps might be correct though (will check soon)

[image]

zaubara, 2023-02-13:
Wasn't able to verify GPU.
Heres another dump while running Cinebench. According to power gadget, CPU should've been 100-108, GPU around 81.
[sensors.txt](https://github.com/exelban/stats/files/10725586/sensors.txt)

exelban, 2023-02-13:
there is no relation between the number of cores and sensors in the apple silicon. So do not expect to see information about 8 cores if you have an 8-core GPU.

tomkoen, 2023-02-14:
I can confirm that CPU sensors values appear to be unusually low in the beta build.

exelban, 2023-02-16:
Hi. Please try this build

[Stats.dmg.zip](https://github.com/exelban/stats/files/10759248/Stats.dmg.zip)

nsklaus, 2023-02-16:
> Hi. Please try this build
> 
> [Stats.dmg.zip](https://github.com/exelban/stats/files/10759248/Stats.dmg.zip)

i still have 6°C displayed with this build.

exelban, 2023-02-16:
yeap, it's how macOS reports that. It's not a bug.

nsklaus, 2023-02-17:
what was the point of this test build then ?

livrth, 2023-03-13:
After I installed stats using homebrew, it doesn't even show CPU or GPU temperature on M2 Pro. But the temperature display on another M2 computer is normal.

[image]

gut4, 2023-03-14:
> After I installed stats using homebrew, it doesn't even show CPU or GPU temperature on M2 Pro. But the temperature display on another M2 computer is normal.
>
The same with m1 macbook. Looks like m1 sensors doesn't supported too.

sumitsahoo, 2023-04-16:
Any way I can help? I have M2 Max.

erikmartinessanches, 2023-06-12:
Sensors do not appear on mac menu even though Sensors button is turned on. M1 Pro, 10 CPU cores, 16 GPU cores, macOS 13.4 (22F66). Stats version 2.8.25. Model identifier: MacBookPro18,3.

xuzhe, 2023-07-30:
if the system reporting wrong temperature, my be the better solution is do not display it to users. Like Parallels did.

https://kb.parallels.com/en/129572

exelban, 2023-08-06:
So, I have spent some time with m2 pro. And I cannot find any differences in the CPU sensors. I have tried different kinds of loads. And didn't notice any specific sensors that related to the efficient or performance cores. As for now, I'm more than sure that CPU cores 1-8 it's for sure the sensors that are related to the CPU. But cannot detect the CPU type between them.

If someone will have any information about m2 series CPU sensors please ping me.

Also, on the m2 platform CPU sensors without load reports 8-10 degrees. So these values are just omitted. The correct value appears only when the system reports them. Even for some sensors it is not 8-10, but 0 degrees for some reason.

closed 2023-08-06 by exelban