exelban/stats #1287: [feature request] add labels in menu bar for sensors and colors for values

nsklaus, 2023-01-29:
**Describe the bug**
in menubar if i add one sensor for cpu temp and another one for gpu temp, there is no label above them, telling which one is which. it's easy to get mistaken, especialy if i toggle sensors on and off while doing testings. depending in the order i do that toggling, the order in menubar can change.. it would be nice if we could have a little label on top of the value to tells to what sensor that value belongs so there can be no errors or confusion because of order changes and such.

since i'm talking about that, it would be nice if temperature value could get color based upon usage.
for example: 
high temp in red ( anything above 90°C)
average temp in yellow (from 60°C up to 90°C)
normal temp in green  (from 0°C to 60°C)

**Details:**
 - Device: macbook pro 2023 (14", M2 pro, 2023)
 - macOS: 13.2
 - Application version: 2.8.5

nsklaus, 2023-01-29:
here's a quick and dirty mockup.
maybe label should reflect it's a sensor too. to differentiate from gpu usage & gpu heat menuitems. adding a unicode glyph:
- thermometer + label for temperature sensors.
- speed meter + label for chip usage

[image]

exelban, 2023-01-31:
It's not possible. There are a lot more sensors that CPU and GPU. Even more, no CPU temperature (available in the macOS). And for each of them, I will need a symbol, and value range for colorizing (ok, it's possible to add).
Additionally, these icons must be drawn with core graphics (I hate to deal with that). 

I would propose using some app that deals only with sensors If you need to visualize a few specific temperatures in the menu bar.

closed 2023-01-31 by exelban

nsklaus, 2023-01-31:
most sensors deal with temperature. so the thermometer icon could be used for all of those. only the label would need to change. the other sensors deal with electricity (amperage, voltage, watts) .. for all those a small electric bolt could be shown. that's just two icons (or unicode glyph.. depending what you would choose).

in this issue i was mainly requesting for labels.
icons and colors are just for eye candy.