exelban/stats #1289: In version 2.8.6 the on/off toogles of the modules are missing on Catalina

irodknocker1, 2023-01-30:
In version 2.8.6 the on/off toogles of the modules are missing on Catalina

[image]

exelban, 2023-01-31:
Does it miss only in the latest update?
Or it's missing after 2.8.0?

irodknocker1, 2023-01-31:
I test today different versions. It worked in v2.7.32 and missing after v2.7.32

exelban, 2023-01-31:
Could you please try this build?

[Stats.dmg.zip](https://github.com/exelban/stats/files/10549240/Stats.dmg.zip)

irodknocker1, 2023-01-31:
It's fixed. Perfect. 

Thanks :)

exelban, 2023-01-31:
ok, thanks. Will be in the next release. Till then you can use that build.

closed 2023-02-05 by exelban