exelban/stats #1290: Bug? Dark Battery Icon

TinyMito, 2023-01-31:
**Describe the bug**
First time trying out the Stats app, I notice the big battery icon in the popup is entirely dark.

This is normal?

[image]

**Details:**
 - Device: [e.g. Macbook Pro 14 (2021]
 - macOS: [e.g. 13.2]
 - Application version: [e.g. 2.8.6]

TinyMito, 2023-01-31:
Update: It appears when the battery is at 100%, it will display dark.

[image]

exelban, 2023-01-31:
it works like that.

closed 2023-01-31 by exelban