exelban/stats #1292: [Feature Request] Temperature decimals

dmlemos, 2023-01-31:
I couldn't find the option to set the temperature reports as decimal. I hope it doesn't come across as nitpick.
The reason I find this important is for benchmarks. A degree of discrepancy makes a big difference, since there is no way to know if the value is off by 0.1 or 0.9.

It should be quite possible to get a more precise temperature and display it, since Macs Fan Control is able to do it.

|Stats|Macs Fan Control
|-|-|
|[image]|[image]|

By the way, being a fan of the software above I got excited when I saw this open-source alternative :)

exelban, 2023-02-05:
Hi. I'm sorry, but there won't be any option to show the decimals in the menu bar. 
I could only propose adding the first decimal to the value in the popup view.

dmlemos, 2023-02-06:
That would be great. Thank you for taking the time

closed 2023-02-12 by exelban