exelban/stats #1297: What is the window names?

theolundqvist, 2023-02-03:
**This is not a bug report**

I am using Yabai which is a tiling window manager. When clicking the battery icon for example the popup gets tiled in with the other windows which is not wanted. To disable managing the popup window I need to know the name of it to add it to the yabai blacklist.

Adding "Battery" or "Stats" does not do the trick.

Thanks

**Details:**
 - Device: [e.g. Macbook Pro 2016]
 - macOS: [e.g. 10.15.5]
 - Application version: [e.g. 2.1.11]

exelban, 2023-02-03:
what do you mean by window name?

theolundqvist, 2023-02-09:
> what do you mean by window name?

Sorry for the lacking description. I guess it is the name that shows upp in the activity bar when an app is focused. By adding "^System Preferences$" to my config file I can turn of window managing for the system preferences app, but Stats don't show upp in the activity bar so I don't know what to write to turn managing off for those windows.

For example:
```yabai -m rule --add app="^Archive Utility$" manage=off```

exelban, 2023-02-19:
Hi. Sorry but I still do not understand what window name you are asking for. Still, there is only one window in the app. Settings window. Please explain exactly what you want to be fixed or added.

luckman212, 2023-02-19:
This is what shows up with UIElementInspector/Xcode Accessibility Inspector, not sure if it's helpful @theolundqvist ...

[image]

[image]

exelban, 2023-02-19:
Thanks. As I can see this window has a title.

closed 2023-02-27 by exelban

theolundqvist, 2023-03-03:
Thanks, it seems to work when I add an exception rule for "Stats", I couldn't get that to work before though.

Mryashbhardwaj, 2023-09-30:
Hi Folks, 
I am facing a similar issue while using stats with [yabai](https://github.com/koekeishiya/yabai).
Fixed the `settings config window` but faced the same issue with widgets. 
@theolundqvist How did you mitigate that?

theolundqvist, 2023-09-30:
@Mryashbhardwaj Currently I am running the following rule: 
```js
yabai -m rule --add app!="^(Arc|Alacritty|Messenger|Messages|Discord|Safari|Spotify)$" manage=off
```
Which floats all windows that are not one of the above. I think that it provides a more predictable environment. I use `option+t` to toggle current window float so it is easy to tile a non managed window.

Mryashbhardwaj, 2023-10-01:
Thanks for the quick response, 
Yeah that seems like a better solution.