exelban/stats #1300: Sony WH-XM4 does not show up in bluetooth battery status menu despite MacOS displaying it.

busti, 2023-02-04:
I have a Sony WH-XM4 headset which I use on my Macbook Pro M1. 
I would like to add it's battery status to the menubar, which I can check in the bluetooth preferences. However, it is not listed in the menu for that item.  

[image]

If there is any further information I can supply please let me know.

exelban, 2023-02-07:
Hi. Could you please provide the output of `/usr/sbin/system_profiler SPBluetoothDataType`?

closed 2023-02-12 by exelban

busti, 2023-02-14:
I apologize. Here is it again with them connected:  
Bluetooth:

      Bluetooth Controller:
          Address: F4:D4:88:86:74:12
          State: On
          Chipset: BCM_4387
          Discoverable: Off
          Firmware Version: 19.6.451.5119
          Product ID: 0x0001
          Supported services: 0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
          Transport: PCIe
          Vendor ID: 0x004C (Apple)
      Connected:
          WH-1000XM4:
              Address: 94:DB:56:8E:0E:C8
              Vendor ID: 0x054C
              Product ID: 0x0D58
              Battery Level: 100 %
              Firmware Version: 2.5.0
              Minor Type: Headset
              Services: 0x800019 < HFP AVRCP A2DP ACL >
      Not Connected:
          Air75 BT5.0 :
              Address: CE:35:8D:A4:BE:FB
              Minor Type: Keyboard
          AirPods Pro:
              Address: 2C:76:00:B7:67:4D
              Vendor ID: 0x004C
              Product ID: 0x2014
              Firmware Version: 5B58
              Minor Type: Headphones
              Serial Number: J3KF14R7F5
          Arc Touch Mouse SE:
              Address: F8:BD:DB:93:CA:AF
              Minor Type: Mouse

busti, 2023-02-14:
And after using them for a while:  
```
Bluetooth:

      Bluetooth Controller:
          Address: F4:D4:88:86:74:12
          State: On
          Chipset: BCM_4387
          Discoverable: Off
          Firmware Version: 19.6.451.5119
          Product ID: 0x0001
          Supported services: 0x382039 < HFP AVRCP A2DP HID Braille AACP GATT Serial >
          Transport: PCIe
          Vendor ID: 0x004C (Apple)
      Connected:
          WH-1000XM4:
              Address: 94:DB:56:8E:0E:C8
              Vendor ID: 0x054C
              Product ID: 0x0D58
              Firmware Version: 2.5.0
              Minor Type: Headset
              Services: 0x800019 < HFP AVRCP A2DP ACL >
      Not Connected:
          Air75 BT5.0 :
              Address: CE:35:8D:A4:BE:FB
              Minor Type: Keyboard
          AirPods Pro:
              Address: 2C:76:00:B7:67:4D
              Vendor ID: 0x004C
              Product ID: 0x2014
              Firmware Version: 5B58
              Minor Type: Headphones
              Serial Number: J3KF14R7F5
          Arc Touch Mouse SE:
              Address: F8:BD:DB:93:CA:AF
              Minor Type: Mouse
```

System dialogue now shows 90%:  
[image]

busti, 2023-02-14:
After searching around for a bit I came across this stackoverflow post: https://apple.stackexchange.com/questions/215256/check-the-battery-level-of-connected-bluetooth-headphones-from-the-command-line  

However, I was not able to extract any meaningful data from `ioreg` using any of the commands in the above post, or manually by scouring through the output of `ioreg -l` using a text editor.

The battery percentage shown by the system has since gone down by another 10% as if to taunt me:  
[image]