exelban/stats #1305: Bug which causes some widgets (and other functions like combined modules) to not work at all.

drcs68, 2023-02-06:
So I have been using stats for the past 2 days (I only use 2 widgets one for Ram and one for Network) and all of a sudden today the Ram is not showing whatever I try (Restart mac, installing new version of stats, etc..). I want to mention that I had the combined module enabled and Ram and Network widgets worked fine together, but now if combined widgets is enabled all widgets turn off, but when I disable it **ONLY** the network widget comes back (this is why I said Ram widget is no longer working whatever I try)

Oh and btw, the program does not launch automatically after boot, even if I add it manually to login items... How can I fix this? I even have 'start at login' turned on.

exelban, 2023-02-06:
Could you please explain step by step what you do?
1. Widgets do not work? It's not possible if you have disabled the combined modules feature.
2. Launch on the boot could work only if you not manipulating the auto login items. The app cannot control the auto launch otherwise.

drcs68, 2023-02-06:
> Could you please explain step by step what you do?
> 
> 1. Widgets do not work? It's not possible if you have disabled the combined modules feature.
> 2. Launch on the boot could work only if you not manipulating the auto login items. The app cannot control the auto launch otherwise.
> 
> Also will be great to share some screenshots or videos of what exactly you think does not work.

Please check this imgur link for more details: https://imgur.com/a/lNjH3NT

drcs68, 2023-02-06:
I just tried resetting all settings and starting all over again. Now whichever module and widget I have enabled, no widget is showing in the menu bar.

exelban, 2023-02-06:
em, please disable all apps in the menu bar. Looks like you don't have enough space to fit all items.

drcs68, 2023-02-07:
you're right, I think it was doing that because there was no space. I managed to make it appear and all is fine so far.

closed 2023-02-07 by exelban