exelban/stats #1307: Network widget not showing unless toggled

PythonCoderAS, 2023-02-07:
When I restart my computer, the network widget (my only widget) does not show up. I have to open up the app settings and turn off the network widget and turn it back on to get it to come back. However, this is not permanent, and on the next restart I have the same issue.

exelban, 2023-02-07:
What widget do you use?

PythonCoderAS, 2023-02-07:
This one: 
[image]

PythonCoderAS, 2023-02-07:
[image]
[image]
[image]
[image]

exelban, 2023-02-07:
Hmm, and what about other modules?
Do they run normally after restarting?

PythonCoderAS, 2023-02-07:
I am not using any other modules.

PythonCoderAS, 2023-02-08:
Ok the problem seems to have fixed itself, maybe after I pressed "check for updates" and it updated itself.

closed 2023-02-08 by PythonCoderAS