exelban/stats #1313: [FR] Add text labels beneath CPU usage bars in popup

luckman212, 2023-02-10:
#### Summary

It would be nice (imo) if there were small labels beneath the CPU core usage levels in the popup to easily identify which are the P/E cores. Example below.

#### Mockup

[image]

exelban, 2023-02-19:
Hi. Thanks for the feedback.
Nope, I'm sure I will not add a description like this. First of all these charts were just a quick solution to my needs. 
I don't think taking space for a text that tells nothing is a good idea. Maybe I will add a tooltip on hover. But not in the near future because it requires quite a lot of work.

closed 2023-02-19 by exelban

luckman212, 2023-02-19:
@exelban Well I can't force you but, a little disappointed that you're abruptly closing without even letting people vote for if they'd find this useful. I see a couple of people upvoted it already, so apparently I'm not the only one who has difficulty remembering which color represents which core.

I'm sponsoring this project to support you and would appreciate at least leaving it open to consideration as a future toggle.

exelban, 2023-02-19:
First of all. Did you read my answer? Especially the part where I say that I will add a tooltip that will indicate the type of core.
Second. Please consider stopping sponsoring me if you expect that I will implement the features that you "want". I don't see any reason to leave this issue open forever because I already know that I will not implement that like that.

PS: there is only one way how Stats represents different core types: first is efficiency, and after is performance.

luckman212, 2023-02-19:
🤷‍♂️  No offense was intended! Yes I did read your reply, I was trying to let you know that I agree, a tooltip would be a fine solution.

I won't stop sponsoring, because I do love Stats and this is a minor thing. It was just a suggestion. If you don't want to do it, I think I already said _"I can't force you"_ and no, of course I don't expect you to implement every feature I want for pennies a day.

I hope it's clear from my participation over the years that I am very appreciative of the work you've done here and this amazing app which you give away. But again, _Thank you_.

[image]

[image]

[image]

[image]

[image]

[image]

etienne07C1, 2023-02-20:
Same thought... we all agree this tool is amazing. 
We just note FR are immediately closed (as mine was) which can look a bit harsh...

luckman212, 2023-02-26:
@exelban Got the new 2.8.10 version, thanks for the E/P custom colors, nice solution!