exelban/stats #1317: Sensors Module Settings to hide/show items don't seem to work

blackketter, 2023-02-13:
**Describe the bug**
I'm trying to slim down the temperatures visible in the Sensors Module.  There are settings for the individual temperatures, but when I turn them off, they stay on in the menu.

In fact, the HID Sensors and Unknown Sensors are the only hide/show controls that work for me.
[image]

**Details:**
 - Device: M2 MacBook Air
 - macOS: 13.2
 - Application version: 2.8.8

exelban, 2023-02-13:
it disables the sensors in the menu bar

blackketter, 2023-02-13:
Ah, I see.  Sorry about that.  

I was confused by the HID Sensors and Unknown Sensors actually working in the window.  I guess this would be a feature request to adjust what's displayed in the window.

exelban, 2023-02-13:
yes. because it enables another type of sensor. Please take a look at the opened issues.