exelban/stats #1318: [Feat] stats of total disk write

etienne07C1, 2023-02-18:
As a M1 mac user, I'd like to estimate SSD write usage depending on my activity.
More specifically seeing how much data has been written in the last hour, over the last days  and when usage is exceeding a given threshold. This in order to prevent SSD wearing.
I thought about something like a bar chart, with days or hours as time range

Currently the only way to do it, is looking at activity monitor, write down the amount of data written to the disk and  divide by the number of days the computer has been on since last boot (which usually gives me around 30-40 GB per day).

Does someone know how to get the amount of data written to disk (as displayed in activity monitor) from command line ?
This would allow to write a script to watch disk usage and outputs stats.

exelban, 2023-02-19:
Hi. Sorry but it's not possible.

closed 2023-02-19 by exelban

etienne07C1, 2023-02-20:
Hi. Do you mean we don't have a way to access that information (e.g. written data since last boot) or for the feat request?
I'm looking for a scripting way to get the written data value shown in activity monitor but I didn't find much information on how to do it on mac so I thought asking here would be the most appropriate place. 
If this can't be added to this tool, I would just be interested in knowing if it is possible to do and how I can do it.
Thanks